intel/linux-sgx-driver #25: Hi question about sgx setting

leesera, 2017-04-30:
Hi, I have a trouble to install isgx module.

My host machine spec is 
`cpu : i7-6700 , mainboard : X11SSZ-QF (supporting SGX)`
And I try to run sgx application through ubuntu 16.06 LTS with virtualbox. 

I put commands for installing sgx driver `isgx`
But the output of command `manprobe` is following

[image]

And the output of `dmesg` after that command is 
[image]

I found the reason of that situation that [boot_cpu_has(X86_FEATURE_SGX)](https://github.com/jethrogb/sgx-utils/blob/master/isgx/isgx_main.c#L105l) returns false.

I do not know why that function does not return well:(
I already set `sgx bios setting` to be enabled

sergeay, 2017-04-30:
The processor you have should have SGX but
it seems SGX has not been enabled.

To make sure of this, have you got the possibility to run cpuid with leaf 18 and sub-leaf 0.
The resulting eax should have the least significant bit equal to 1. If not SGX was not enabled by bios.

leesera, 2017-04-30:
@sergeay thank you for replying my question:)
I already set my host machine to be SGX enabled by bios setting. (Window is running on my machine)
But it does not work in ubuntu which is running on virtualbox. Is there anything to set more?

sergeay, 2017-04-30:
I got it.
You are using Ubuntu as a guest machine from a windows host machine.
SGX in a guest machine is not supported yet.
If you want Ubuntu to work with SGX, you will need to install it natively (bare metal).

leesera, 2017-04-30:
@sergeay I understand what you mean. Thank you for helping me!

closed 2017-04-30 by leesera