intel/linux-sgx-driver #35: Exec format error when adding isgx module to updated Linux kernel 4.4.0-92

cho0thee, 2017-08-18:
When trying to insert the isgx module into the Linux kernel 4.4.0-92 the following error message is shown

```
$ sudo /sbin/modprobe -v isgx
insmod /lib/modules/4.4.0-92-generic/kernel/drivers/intel/sgx/isgx.ko 
modprobe: ERROR: could not insert 'isgx': Exec format error
```

I am using Ubuntu 16.04 LTS 64-bit on an Intel Core i5-6440HQ CPU and updated to kernel version 4.4.0-92 (from the Ubuntu repositories) a few days ago. When booting into the kernel version 4.4.0-79, the module is inserted from /etc/modules and usable without any problems.

sergeay, 2017-08-20:
It seems you need to reinstall the driver since you changed the kernel.
So I would remove the dependent components (probably psw and sdk) and then the driver and re-install the chain.

cho0thee, 2017-08-21:
I must have missed a step there, before. Thank you

closed 2017-08-21 by cho0thee