intel/linux-sgx-driver #36: sgx_ioc_enclave_add_page is interrupted and returns -EINVAL

deeglaze, 2017-09-07:
About 0.1% of the time, we get an add page failure while loading our 15MiB enclave.
Instrumenting the driver, I found that `sgx_page_alloc(0)` fails here https://github.com/01org/linux-sgx-driver/blob/master/sgx_ioctl.c#L413
Which, digging further I found is due to `signal_pending` sending back `ERR_PTR(-ERESTARTSYS)`. Could this be a bug in the driver since it's returned as -EINVAL at https://github.com/01org/linux-sgx-driver/blob/master/sgx_ioctl.c#L700 ?

jsakkine-intel, 2017-09-08:
Not related to bug itself but sgx_page_alloc() should return `-EINTR` (note to self mainly) and call site should decide how to deal with it.

In the case of `sgx_ioc_enclave_create()` the return value should be checked and if it is `-EINTR` it should return `-ERESTARTSYS`.

It's a driver bug. Returning `-EINVAL` makes zero sense.

haitaohuang, 2017-09-08:
I think there are more cases where sgx_init_page/sgx_alloc_page used and possibly return -ERESTARTSYS, which should have returned -EINTR. And callers of those functions should handle properly. That may be the reason why you see 1% failures in create_enclave

jsakkine-intel, 2017-09-08:
Sure, I'll go through the call sites.

jsakkine-intel, 2017-09-08:
Now you are mixing two things. The regression here is that the call site returns `-EINVAL`. The comment about `-EINTR` was completely unrelated to the regression.

haitaohuang, 2017-09-08:
here is an attempt to track down those pending signal situations, where -EINTR returned from either mmu_notifier_register or sgx_alloc_page(0):
https://github.com/haitaohuang/linux-sgx-driver/commit/f2ba8fbb779b7e26887ae2b8bbec04426ccb74a0
Please review or give it a try.

haitaohuang, 2017-09-08:
should einit IOC also return -ERESTARTSYS at https://github.com/01org/linux-sgx-driver/blob/master/sgx_ioctl.c#L849?
It is idempotent, isn't it?

jsakkine-intel, 2017-09-08:
I don't know how this relates to the bug in question.

haitaohuang, 2017-09-12:
Yes, we won't be sure to find fix of the original failures until we can reproduce. The patch only attempt to fix  those EINTR cases in driver, which may  help application side to isolate the issue.

deeglaze, 2017-09-12:
Your patch worked for me, @haitaohuang, except now I hit yet another problem that appears unrelated since it's in the uae service. [I've opened an issue in linux-sgx.](https://github.com/01org/linux-sgx/issues/158)

haitaohuang, 2018-01-17:
fixed

closed 2018-01-17 by haitaohuang