intel/linux-sgx-driver #50: Compilation error in 4.14: "impossible constraint in asm"

chrisr3, 2018-01-17:
Trying to compile the latest `isgx` module from the master branch:
```
$ make
make -C /lib/modules/4.14.13-300.fc27.x86_64/build SUBDIRS=/home/chris/workspace/linux-sgx-driver CFLAGS_MODULE="-DDEBUG -g -O0" modules
make[1]: Entering directory '/usr/src/kernels/4.14.13-300.fc27.x86_64'
arch/x86/Makefile:244: CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended.
  CC [M]  /home/chris/workspace/linux-sgx-driver/sgx_util.o
In file included from ./include/linux/compiler_types.h:58:0,
                 from ./include/uapi/linux/stddef.h:2,
                 from ./include/linux/stddef.h:5,
                 from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/types.h:6,
                 from /home/chris/workspace/linux-sgx-driver/sgx_arch.h:57,
                 from /home/chris/workspace/linux-sgx-driver/sgx_asm.h:61,
                 from /home/chris/workspace/linux-sgx-driver/sgx.h:63,
                 from /home/chris/workspace/linux-sgx-driver/sgx_util.c:61:
./arch/x86/include/asm/jump_label.h: In function ‘put_page’:
./include/linux/compiler-gcc.h:276:38: warning: asm operand 0 probably doesn’t match constraints
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/jump_label.h:36:2: note: in expansion of macro ‘asm_volatile_goto’
  asm_volatile_goto("1:"
  ^~~~~~~~~~~~~~~~~
./include/linux/compiler-gcc.h:276:38: warning: asm operand 1 probably doesn’t match constraints
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/jump_label.h:36:2: note: in expansion of macro ‘asm_volatile_goto’
  asm_volatile_goto("1:"
  ^~~~~~~~~~~~~~~~~
./include/linux/compiler-gcc.h:276:38: error: impossible constraint in ‘asm’
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/jump_label.h:36:2: note: in expansion of macro ‘asm_volatile_goto’
  asm_volatile_goto("1:"
  ^~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:315: /home/chris/workspace/linux-sgx-driver/sgx_util.o] Error 1
make[1]: *** [Makefile:1511: _module_/home/chris/workspace/linux-sgx-driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.14.13-300.fc27.x86_64'
make: *** [Makefile:16: default] Error 2
```
I see this error on both Fedora 27 and Arch Linux.

chrisr3, 2018-01-17:
Assembler version:
```bash
$ as -V
GNU assembler version 2.29.1 (x86_64-pc-linux-gnu) using BFD version (GNU Binutils) 2.29.1
```

chrisr3, 2018-01-17:
This seems to be a recurrence of #13, and even has the same workaround.

sergeay, 2018-01-21:
So I understand that the only difference is the addition of M=$(PWD) instead of SUBDIRS=$(PWD) in the original Makefile.
Can we close it?

chrisr3, 2018-01-22:
I have been working around this problem by removing the `-O0` option from `CFLAGS_MODULE` in the `Makefile`. I would consider closing it without an attempt at understanding it to be premature.

hecto2, 2018-02-03:
I ran into similiar problems when compiling kernel 4.14 by changing -O2 to -O0.

`In file included from ./include/linux/compiler_types.h:58:0,
                 from ./include/linux/compiler.h:5,
                 from ./arch/x86/include/asm/atomic.h:5,
                 from ./include/linux/atomic.h:5,
                 from ./include/linux/crypto.h:20,
                 from arch/x86/kernel/asm-offsets.c:9:
./arch/x86/include/asm/cpufeature.h: In function ‘_static_cpu_has’:
./include/linux/compiler-gcc.h:270:38: warning: asm operand 0 probably doesn’t match constraints
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/cpufeature.h:148:3: note: in expansion of macro ‘asm_volatile_goto’
   asm_volatile_goto("1: jmp 6f\n"
   ^
./include/linux/compiler-gcc.h:270:38: warning: asm operand 2 probably doesn’t match constraints
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/cpufeature.h:148:3: note: in expansion of macro ‘asm_volatile_goto’
   asm_volatile_goto("1: jmp 6f\n"

./include/linux/compiler-gcc.h:270:38: error: impossible constraint in ‘asm’
 #define asm_volatile_goto(x...) do { asm goto(x); asm (""); } while (0)
                                      ^
./arch/x86/include/asm/cpufeature.h:148:3: note: in expansion of macro ‘asm_volatile_goto’
   asm_volatile_goto("1: jmp 6f\n"
   ^`

The trouble above was caused by extended assembly in arch/x86/include/asm/cpufeature.h, where the modifiers of the operand 0 and 2 were i, meaning that immediate integers were expected. But they both used parameter bit, which was a variable instead of a constant. If -O2 was specified, the inline function _static_cpu_has was expanded with a constant in place of parameter bit, which bypassed the restriction.

`static __always_inline __pure bool _static_cpu_has(u16 bit)
{
                asm_volatile_goto("1: jmp 6f\n"
                         "2:\n"
                         ".skip -(((5f-4f) - (2b-1b)) > 0) * "
                                 "((5f-4f) - (2b-1b)),0x90\n"
                         "3:\n"
                         ".section .altinstructions,\"a\"\n"
                         " .long 1b - .\n"              /* src offset */
                         " .long 4f - .\n"              /* repl offset */
                         " .word %P1\n"                 /* always replace */
                         " .byte 3b - 1b\n"             /* src len */
                         " .byte 5f - 4f\n"             /* repl len */
                         " .byte 3b - 2b\n"             /* pad len */
                         ".previous\n"
                         ".section .altinstr_replacement,\"ax\"\n"
                         "4: jmp %l[t_no]\n"
                         "5:\n"
                         ".previous\n"
                         ".section .altinstructions,\"a\"\n"
                         " .long 1b - .\n"              /* src offset */
                         " .long 0\n"                   /* no replacement */
                         " .word %P0\n"                 /* feature bit */
                         " .byte 3b - 1b\n"             /* src len */
                         " .byte 0\n"                   /* repl len */
                         " .byte 0\n"                   /* pad len */
                         ".previous\n"
                         ".section .altinstr_aux,\"ax\"\n"
                         "6:\n"
                         " testb %[bitnum],%[cap_byte]\n"
                         " jnz %l[t_yes]\n"
                         " jmp %l[t_no]\n"
                         ".previous\n"
                         : : **"i" (bit)**, "i" (X86_FEATURE_ALWAYS),
                             **[bitnum] "i" (1 << (bit & 7)),**
                             [cap_byte] "m" (((const char *)boot_cpu_data.x86_capability)[bit >> 3])
                         : : t_yes, t_no);
        t_yes:
                return true;
        t_no:
                return false;
}`

kaimast, 2018-04-09:
Same problem on Ubuntu 18.04 with linux 4.15

sergeay, 2018-04-24:
The issue has to do with conflicting flags - I will submit a change to the Makefile.
Thanks for raising this issue.

sergeay, 2018-05-29:
closing issue - fix was submitted.

closed 2018-05-29 by sergeay