intel/linux-sgx-driver #53: Modprobe fail when using fedora/atomic host

GabrielSVinha, 2018-01-25:
I've been trying to run the build in a fedora virtual machine hosted by ubuntu and meet the following error:

`modprobe: ERROR: could not insert 'isgx': Exec format error`

My guess was that there was an error while cross-compiling, then found this issue. Is this the right place? Can anyone assist me?

This was mentioned in https://github.com/intel/linux-sgx/issues/36.

andyzyb, 2018-01-25:
Can you elaborate the process you build and load the driver?   Which OS to build and which to load?  Typically you have to build and run on the same kernel version.

BTW, if you modprobe (load) the driver in a virtual machine, you need to make sure the VMM does support SGX.

GabrielSVinha, 2018-01-25:
I am building in a fedora-25 machine with chroot from a Ubuntu 16.04. I am using the kernel-headers from the fedora system. Also, SGX is supported from the hypervisor (VMM).

closed 2020-04-24 by haitaohuang