intel/linux-sgx-driver #59: Build error: stack-protector enabled but compiler support broken

falkenteich, 2018-04-09:
I'm trying to install the SGX Driver on RHEL7.4-64bit.

I first downloaded the SGX Driver installer from https://01.org/intel-software-guard-extensions/downloads, but I got the following error:

    # ./sgx_linux_x64_driver_1bf506e.bin
    Unpacking Intel SGX Driver ... done.
    Verifying the integrity of the install package ... done.
    Installing Intel SGX Driver ...
    /tmp/sgx-driver-Jb6YTw ~/sgxdownloads
    install -d /opt/intel/sgxdriver/package
    install -d /opt/intel/sgxdriver/scripts
    install package/* /opt/intel/sgxdriver/package
    install scripts/* /opt/intel/sgxdriver/scripts
    ~/sgxdownloads
    /opt/intel/sgxdriver/package ~/sgxdownloads
    make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build SUBDIRS=/opt/intel/sgxdriver/package CFLAGS_MODULE="-DDEBUG -g -O0" modules
    make[1]: Entering directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    arch/x86/Makefile:96: stack-protector enabled but compiler support broken
    arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended..  Stop.
    make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    make: *** [default] Error 2

Then I tried to download and build the SGX Driver directly from this Git repo, but I got the same error:

    # make
    make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build SUBDIRS=/root/sgxdownloads/linux-sgx-driver-master CFLAGS_MODULE="-DDEBUG -g -O0" modules
    make[1]: Entering directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    arch/x86/Makefile:96: stack-protector enabled but compiler support broken
    arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended..  Stop.
    make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    make: *** [default] Error 2

After reading through the repo issues, I tried tweaking the Makefile, first by changing SUBDIRS=$(PWD) to M=$(PWD), and then by removing the -O0 option from the CFLAGS_MODULE.
Neither of those, nor both of those, improved the situation. Here's the output with both changes:

    # make
    make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build M=/root/sgxdownloads/linux-sgx-driver-master CFLAGS_MODULE="-DDEBUG -g" modules
    make[1]: Entering directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    arch/x86/Makefile:96: stack-protector enabled but compiler support broken
    arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended..  Stop.
    make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
    make: *** [default] Error 2

Can anyone show me the errors of my ways?  Thanks.

sergeay, 2018-04-10:
You seem to have CONFIG_RETPOLINE=y. Where does this come from?
As far as I remember, it was introduced in kernel 4.16 but your Redhat platform has 3.10...
So this is weird.
Anyway if CONFIG_RETPOLINE=y then you may need the latest gcc version since older versions do not support it.
Again, something looks not standard in your platform as RETPOLINE was introduced in latest kernels which is not your case.
This problem has nothing to do with building the SGX driver and I would expect any kernel module build to hit the same issue.

sergeay, 2018-04-26:
Build errors for >= kernel 4.15 have been fixed.

jacmarjorie, 2018-05-02:
Is there a fix for this? I'm hitting the same issue with gcc 7.2.1 via devtoolset-7 and same kernel (on centos7):

```bash
-bash-4.2$ make
make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build SUBDIRS=/nfs/home/jaclyns/ati/linux-sgx-driver modules
make[1]: Entering directory '/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended..  Stop.
make[1]: Leaving directory '/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
make: *** [Makefile:15: default] Error 2
```

sergeay, 2018-05-02:
CONFIG_RETPOLINE=y requires gcc version 7.3 or higher.
You probably have a older version, so check your version and upgrade gcc.

sergeay, 2018-05-29:
Can this one be closed, was your problem resolved?

closed 2018-05-29 by sergeay