intel/linux-sgx-driver #63: Compilation error with kernel 4.13.0-38-generic:  fatal error: asm/msr-index.h: No such file or directory

HolmesSherlock, 2018-05-28:
As Linux kernel has [stopped exporting](https://github.com/torvalds/linux/commit/25dc1d6cc3082aab293e5dad47623b550f7ddd2a) `asm/msr-index.h` to user-land starting kernel version `4.12-rc1` , the build fails with the following error message: `fatal error: asm/msr-index.h: No such file or directory`

The build system configuration is as follows:
```
$ uname -r
4.13.0-38-generic
$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"
```

sergeay, 2018-05-29:
It says: "x86: stop exporting msr-index.h to userland"

my Ubuntu 16.04 has the following:
$ uname -r
4.13.0-43-generic

and msr-index.h is at:
$ sudo find / -name 'msr-index.h'
/usr/src/linux-headers-4.13.0-43/arch/x86/include/asm/msr-index.h

And the driver compilation works fine

HolmesSherlock, 2018-05-29:
Extremely sorry. My host had `ARCH` and `CROSS_COMPILE` environment variables wrongly set from a previous kernel build. Closing this issue.

closed 2018-05-29 by HolmesSherlock