intel/linux-sgx-driver #68: no /dev/isgx after installing driver

pblin, 2018-10-12:
Azure VM, OS: Ubuntu 16.04
No /dev/isgx after installation.

piotr-roslaniec, 2018-12-30:
I also faced this issue on Azure Confidential Computing VM (Standard_DC2s). 
It was resolved by upgrade to 18.04 suggested by @pblin 

```
$  lsb_release -a 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic
$ uname -r 
4.15.0-1036-azure
```

stefandeml, 2019-01-13:
I'm still facing the issue under Ubuntu 18.04:
`
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic
`
I have neither /dev/sgx nor dev/isgx available on my machine. 
The aesmd service works, as do the SampleCodes.

Any idea how to debug this?

clefru, 2019-01-13:
In this git repo run: 
```
make install
depmod -a
```
https://github.com/intel/linux-sgx-driver/pull/75 adds the depmod step to make install.

stefandeml, 2019-01-13:
Thanks for the support.
Unfortunately this fix didn't help on my end. Still missing `/dev/isgx`

sergeay, 2019-01-14:
What is the output of dmesg | grep sgx?

stefandeml, 2019-01-21:
Can be closed. I had a BIOS reset and SGX has been disabled .

danintel, 2019-07-18:
Is `/dev/isgx` upward compatible with `/dev/sgx` ?
That is, I can just switch out `/dev/sgx` with `/dev/isgx` and everything should continue to work OK?

closed 2020-03-03 by haitaohuang