intel/linux-sgx-driver #97: build error: code model kernel does not support PIC mode

NanFeng009, 2019-08-01:
make -C /lib/modules/4.15.0-29-generic/build SUBDIRS=/home/sgx/linux-sgx-driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-29-generic'
  CC [M]  /home/sgx/linux-sgx-driver/sgx_main.o
cc1: error: code model kernel does not support PIC mode
scripts/Makefile.build:332: recipe for target '/home/sgx/linux-sgx-driver/sgx_main.o' failed
make[2]: *** [/home/sgx/linux-sgx-driver/sgx_main.o] Error 1
Makefile:1552: recipe for target '_module_/home/sgx/linux-sgx-driver' failed
make[1]: *** [_module_/home/sgx/linux-sgx-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-29-generic'
Makefile:15: recipe for target 'default' failed
make: *** [default] Error 2

-------
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)

sergeay, 2019-08-01:
How did you set PIC mode for your kernel?

sergeay, 2019-08-01:
We do not get those even on 19.04
Would adding the following in the Makefile help?
KBUILD_CFLAGS += -fno-pie

NanFeng009, 2019-08-05:
>How did you set PIC mode for your kernel?
No special setting, just a new install ubuntu with default setting. but ubuntu 1804 was installed in a hyper-v on windows 10. 
>Would adding the following in the Makefile help?
>>KBUILD_CFLAGS += -fno-pie
no help, same error

ZongminGu, 2019-08-05:
Below link may help you:
https://askubuntu.com/questions/851433/kernel-doesnt-support-pic-mode-for-compiling

NanFeng009, 2019-08-05:
>https://askubuntu.com/questions/851433/kernel-doesnt-support-pic-mode-for-compiling
for this defect,  "no longer affects: | gcc-defaults (Ubuntu Xenial)" and here the ubutu is 1804 and gcc version is 7.0.4, the defect should be no long exist.
Try adding some *cflags" does not help.

zavalyshyn, 2020-03-09:
So I had the same issue while installing the latest 2.6 version of the driver on Ubuntu Server 18.04.4. Adding flags did help to pass through the original error message but then another popped up (asm related).

So then I thought maybe it's a problem of the 2.6 version. So I switched the branch to `sgx_driver_2.0` tag, made a fresh `make clean` and `make` again. And it compiled successfully. Interestingly, right after that I switched back to sgx_driver_2.6 tag and tried to compile again. It worked this time.
I have no idea what was the problem before.

closed 2020-04-24 by haitaohuang