intel/linux-sgx-driver #101: run `sudo /sbin/modprobe isgx` error

jithendaraa, 2019-09-21:
run `sudo /sbin/modprobe isgx` error
modprobe: ERROR: could not insert 'isgx': Operation not permitted
Can anyone help me with this? Why am I getting this error and how to proceed

Also after `sudo apt-get install linux-headers-$(uname -r)` I get the following error

**Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.0.0-27-generic is already the newest version (5.0.0-27.28).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up intel-sgx-dkms (1.9-2) ...
Removing old intel-sgx-1.9 DKMS files...

------------------------------
Deleting module version: 1.9
completely from the DKMS tree.
------------------------------
Done.
Loading new intel-sgx-1.9 DKMS files...
Building for 5.0.0-27-generic
Building initial module for 5.0.0-27-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/intel-sgx-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.0.0-27-generic (x86_64)
Consult /var/lib/dkms/intel-sgx/1.9/build/make.log for more information.
dpkg: error processing package intel-sgx-dkms (--configure):
 installed intel-sgx-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 intel-sgx-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)**

DanistyWuKids, 2019-10-02:
I got the same error, but if I go straight download the prebuild binary file from 01.org, it will work. But I still cannot compile it with Hardware mode (i5 7200U on Ubuntu 18.04)
`$ sudo /sbin/modprobe isgx`

> modprobe: ERROR: could not insert 'isgx': Operation not permitted

`$ sudo apt-get install linux-headers-$(uname -r)`

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> linux-headers-4.18.0-15-generic is already the newest version (4.18.0-15.16~18.04.1).
> linux-headers-4.18.0-15-generic set to manually installed.
> 0 to upgrade, 0 to newly install, 0 to remove and 8 not to upgrade.

DanistyWuKids, 2019-10-02:
Oh, after I do some research, I have solved this to me.
 @jithendaraa I think you have enabled the UEFI Secure Boot on your machine.
Refer to Ubuntu Wiki about [SecureBoot](https://wiki.ubuntu.com/UEFI/SecureBoot), then I got knowing there have 2 files (shimx64.efi and grubx64.efi). First one is for Parallel Boot Ubuntu after SecureBoot enabled in your Firmware. The second one is the regular one.

Therefore, I have referred to this page of [signing the isgx by yourself](https://askubuntu.com/questions/760671/could-not-load-vboxdrv-after-upgrade-to-ubuntu-16-04-and-i-want-to-keep-secur/768310#768310). So for example, in step 2 should be 
`sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./MOK.priv ./MOK.der $(modinfo -n isgx)` after all done and enroll the MOK successfully, redo the installation and this command can be executed without error.

wuliangshun, 2021-01-12:
"modprobe: ERROR: could not insert 'isgx': Exec format error"
when I execute "sudo /sbin/modprobe isgx" I got the above error, how do I fix it?  My PC is equipped with i5-10210U installed Ubuntu 18.04.

Here is the output after run "make":
_make -C /lib/modules/5.4.0-60-generic/build M=/home/wuliangshun/SGX/driver/linux-sgx-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-60-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-60-generic'_

haitaohuang, 2021-01-13:
Exec format error usually happens when you build with one kernel version and install it on a different kernel. Make sure you are doing a clean build

closed 2022-06-13 by haitaohuang

lim-yoona, 2023-12-07:
> Exec format error usually happens when you build with one kernel version and install it on a different kernel. Make sure you are doing a clean build

Hi! I also meet this problem. How can I ensure that the driver is build and installed with the same kernel?