intel/linux-sgx-driver #108: modprobe: ERROR: could not insert 'isgx': Operation not permitted under CentOS 7x64

kvishnivetsky, 2019-11-15:
Hi, There!

We have trouble in running kernel module for SGX support.

Intel(R) Xeon(R) CPU E5-2630 v4 @ 2.20GHz (fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu cpuid_faulting pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb cat_l3 cdp_l3 intel_ppin intel_pt ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdt_a rdseed adx smap xsaveopt cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts md_clear spec_ctrl intel_stibp flush_l1d)

CentOS Linux release 7.7.1908 (Core)

Linux d214.sancom.in 3.10.0 #1 SMP Thu Jun 27 15:10:55 MSK 2019 x86_64 x86_64 x86_64 GNU/Linux

Make goes perfect, but modprobe/insmod fail.

_modprobe: ERROR: could not insert 'isgx': Operation not permitted_

andyzyb, 2019-11-18:
Is secure boot enabled on the system?

haitaohuang, 2020-04-24:
close for now. Please reopen if this is still an issue

closed 2020-04-24 by haitaohuang