intel/linux-sgx-driver #110: Not able to run SampleEnclave because of SGX module seems to be disabled in BIOS while my BIOS tells it is enabled

kleirama, 2020-02-25:
I have installed SGX SDK version 2.8, SGX PSW and SGX DRIVER version 2.6 downloaded by this site [](https://download.01.org/intel-sgx/sgx-linux/2.8/distro/ubuntu18.04-server/) and installed them using this guide [](https://download.01.org/intel-sgx/linux-2.3/docs/Intel_SGX_Installation_Guide_Linux_2.3_Open_Source.pdf) and then when I run SampleEnclave project I am able to build it but when I execute 

> ./app

I got the following error: 

> Info: Please make sure SGX module is enabled in the BIOS, and install SGX driver afterwards.
> Error: Invalid SGX device.
> Enter a character before exit ...

And that is super strange since I have controled several times the BIOS and SGX is enabled. I do not know what is going wrong. Any suggestions?

haitaohuang, 2020-03-03:
do you see /dev/isgx after installing driver?

shao158, 2020-03-12:
I have the same problem running SampleEnclave.

I can see /dev/isgx and "isgx" in /etc/modules. Thanks for any help.

andyzyb, 2020-03-12:
Can you list all the packages you've installed for SGX ?   There might be missing some packages.

shao158, 2020-03-12:
Thanks for replying.

Yes, according to this [page](https://github.com/intel/linux-sgx), I ran the following commands to make sure I have all dependencies:

sudo apt-get install build-essential ocaml ocamlbuild automake autoconf libtool wget python libssl-dev git
sudo apt-get install libssl-dev libcurl4-openssl-dev protobuf-compiler libprotobuf-dev debhelper cmake reprepro
sudo apt-get install build-essential python
sudo apt-get install libssl-dev libcurl4-openssl-dev libprotobuf-dev

shao158, 2020-03-12:
And I installed the following SGX packages:

libsgx-enclave-common
libsgx-epid
libsgx-launch
libsgx-urts
libsgx-uae-service
libsgx-quote-ex

andyzyb, 2020-03-12:
Did you use the Intel Debian repo for installation (apt-get) for SGX packages, then it will resolve all the dependency? 

If you install it through dkpg one by one,  there are many other packages to be installed. 

Can you check if aesm service is running, like "sudo service aesmd status"?

shao158, 2020-03-12:
Thanks for reminding this.
Now I installed aesm, and SampleEnclave is running perfect in both Hardware and Sim modes.

But LocalAttestation still has an error, for both Hardware and Sim modes,
saying "failed to load enclave". Is this also because I missed some packages?
I already used apt-get to check all six packages list above.
And "sudo service aesmd status" is actively running.

andyzyb, 2020-03-12:
For LocalAttestion, there seems an error in README. Can you try below command?

  $ cd SampleCode/LocalAttestation
  $ make	  $ make
  $ **cd bin**
  $ ./app	  $ ./app

shao158, 2020-03-12:
Wow. This works. Thanks you very much.

closed 2020-04-24 by haitaohuang