intel/linux-sgx-driver #128: AESM Service can't access from China

jasl, 2020-12-04:
I don't know where I can push AESM Service's problem, but this emergency and embarrassing

ps.sgx.trustedservices.intel.com can't access from China, now, I believe it affact all SGX apps

[image]

llly, 2020-12-07:
Thanks a lot. I can reproduce this issue. Sometimes accessing this URL shows the unexpected page. Sometimes it returns expected error code 405.

jsun39, 2020-12-07:
We are working to address it. Thanks to report it.

jasl, 2020-12-07:
> Thanks a lot. I can reproduce this issue. Sometimes accessing this URL shows the unexpected page. Sometimes it returns expected error code 405.

It depends on DNS resolve to where

typically, in Beijing

```
dig ps.sgx.trustedservices.intel.com

; <<>> DiG 9.10.6 <<>> ps.sgx.trustedservices.intel.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 50330
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;ps.sgx.trustedservices.intel.com. IN	A

;; ANSWER SECTION:
ps.sgx.trustedservices.intel.com. 862 IN CNAME	ps.sgx1.infra-host.com.
ps.sgx1.infra-host.com.	22	IN	CNAME	prc-ps.sgxprod.adsdcsp.com.
prc-ps.sgxprod.adsdcsp.com. 261	IN	A	40.126.120.63

;; Query time: 63 msec
;; SERVER: 10.0.0.1#53(10.0.0.1)
;; WHEN: Mon Dec 07 14:04:26 CST 2020
;; MSG SIZE  rcvd: 136
```

<ps.sgx.trustedservices.intel.com> will redirect to <prc-ps.sgxprod.adsdcsp.com> you can 100% reproduce the problem.

llly, 2020-12-09:
This issue should be fixed now.

jasl, 2020-12-09:
thank you! I close the issue now

closed 2020-12-09 by jasl

ltinerary, 2021-04-27:
hello @jasl 
When I visit this site（[ps.sgx.trustedservices.intel.com](http://ps.sgx.trustedservices.intel.com/)）, it shows 403

[image]

What is this website used for?
Thanks a lot ~

jasl, 2021-05-01:
> hello @jasl
> When I visit this site（[ps.sgx.trustedservices.intel.com](http://ps.sgx.trustedservices.intel.com/)）, it shows 403
> 
> [image]
> 
> What is this website used for?
> Thanks a lot ~

This is the IAS API site, not use for open in browser

ltinerary, 2021-05-02:
> > hello @jasl
> > When I visit this site（[ps.sgx.trustedservices.intel.com](http://ps.sgx.trustedservices.intel.com/)）, it shows 403
> > [image]
> > What is this website used for?
> > Thanks a lot ~
> 
> This is the IAS API site, not use for open in browser

Thank you！☺