intel/linux-sgx-driver #131: ioctl returns ENOENT on final enclave initialisation

CitadelCore, 2021-02-09:
Opening this issue here because I think it could possibly be an issue with this driver rather than the aesmd daemon. (Relevant issue for that is here: https://github.com/intel/linux-sgx/issues/671)

Through strace I can see all the operations happening successfully. The enclave is created, pages are loaded, it is only at INIT (ioctl operation 0x2) when the driver reports that the enclave does not exist. sgx_encl_find in the driver is what is actually returning ENOENT.

Logs from both the aesmd service and strace are attached. There are no relevant lines in dmesg. I can reproduce the issue on both my systems with Intel CPUs.

[journalctl.log](https://github.com/intel/linux-sgx-driver/files/5949658/journalctl.log)
[trace.log](https://github.com/intel/linux-sgx-driver/files/5949659/trace.log)

CitadelCore, 2021-02-11:
@haitaohuang any idea what's going on here?

haitaohuang, 2021-02-18:
Sorry for late reply. I don't see ENOENT from ioctl. Can you copy the relevant section of the strace? EINIT failure is usually some attribute does not work. If this is nonFLC and you are running provisioning enclave, make sure to use official binaries signed by Intel

haitaohuang, 2021-02-18:
Ah, just saw you copied strace on  intel/linux-sgx#671. I think you mean 2 returned at this line:
ioctl(4, _IOC(_IOC_WRITE, 0xa4, 0x2, 0x18), 0x7ffccd992240) = 2

I believe it's not ENOENT. It's likely the error code returned from EINIT, so likely sigstruct invalid or attributes mismatch. Also notice you built the aesm from source. Did you make sure to use official binaries of QE, PVE, LE if this is on non-FLC?

CitadelCore, 2021-02-18:
I should be using signed binaries - for reference, this is the Nix derivation I wrote to build the SDK and PSW from source: https://gist.github.com/CitadelCore/2acfc56a8abaead11a2d32a785c267b4 - as you can see I'm fetching them from https://download.01.org/intel-sgx/sgx-linux.

Will get back to you re. the output of that debug statement with kernel debugging on.

Time0o, 2022-05-08:
@CitadelCore Did you ever resolve this? I think I'm seeing the same problem [here](https://github.com/intel/linux-sgx/issues/834).