intel/linux-sgx-driver #137: "Failed to load enclave." error with Centos7.8 release

deepakdc01, 2021-08-07:
I created this issue thinking this might be specific to the version I used and the older CentOS7.8 I'm using. Basically, after following the installation guide of version 2.8 (the latest that can run on CentOS 7) and installed all the necessary components - driver, sdk, with the supporting libraries, I get this infamous error when running one of the example.

Here are my environment details:

**[root@localhost sgx-aesm-service]# uname -a**
Linux localhost.localdomain 3.10.0-1160.36.2.el7.x86_64 #1 SMP Wed Jul 21 11:57:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

**[root@localhost sgx-aesm-service]# rpm -qa | grep -i sgx**
libsgx-ae-pce-2.8.100.3-1.el7.centos.x86_64
libsgx-quote-ex-2.8.100.3-1.el7.centos.x86_64
libsgx-launch-2.8.100.3-1.el7.centos.x86_64
libsgx-ae-qe3-1.4.100.3-1.el7.centos.x86_64
libsgx-ae-le-2.8.100.3-1.el7.centos.x86_64
libsgx-enclave-common-2.8.100.3-1.el7.centos.x86_64
libsgx-ae-epid-2.8.100.3-1.el7.centos.x86_64
libsgx-aesm-ecdsa-plugin-2.8.100.3-1.el7.centos.x86_64
sgx-aesm-service-2.8.100.3-1.el7.centos.x86_64
libsgx-urts-2.8.100.3-1.el7.centos.x86_64
libsgx-aesm-epid-plugin-2.8.100.3-1.el7.centos.x86_64
libsgx-aesm-quote-ex-plugin-2.8.100.3-1.el7.centos.x86_64
libsgx-uae-service-2.8.100.3-1.el7.centos.x86_64
libsgx-aesm-launch-plugin-2.8.100.3-1.el7.centos.x86_64
libsgx-epid-2.8.100.3-1.el7.centos.x86_64
libsgx-aesm-pce-plugin-2.8.100.3-1.el7.centos.x86_64

**[root@localhost LocalAttestation]# systemctl status aesmd**
● aesmd.service - Intel(R) Architectural Enclave Service Manager
   Loaded: loaded (/usr/lib/systemd/system/aesmd.service; enabled; vendor preset: disabled)
   Active: active (running) since Sat 2021-08-07 12:01:45 IST; 1h 26min ago
 Main PID: 2276 (aesm_service)
   CGroup: /system.slice/aesmd.service
           └─2276 /opt/intel/sgx-aesm-service/aesm/aesm_service

Aug 07 12:01:40 localhost.localdomain systemd[1]: Starting Intel(R) Architectural Enclave Service Manager...
Aug 07 12:01:45 localhost.localdomain systemd[1]: Started Intel(R) Architectural Enclave Service Manager.
Aug 07 12:01:49 localhost.localdomain aesm_service[2276]: The server sock is 0x22e1a80
Aug 07 12:16:28 localhost.localdomain aesm_service[2276]: InKernel LE loaded

**Downloaded the source code from https://github.com/intel/linux-sgx/tree/sgx_2.8
And ran one of the examples:**
[root@localhost LocalAttestation]# ./bin/app
failed to load enclave.
[root@localhost LocalAttestation]# 

To confirm, I downloaded the sgx-software-enable from https://github.com/intel/sgx-software-enable 
[root@localhost sgx-software-enable]$ ./sgx_enable -s
**Intel SGX is already enabled on this system**
[root@localhost sgx-software-enable]$

haitaohuang, 2021-08-10:
The enclaves for local attestation are built in bin directory. You need cd bin && ./app

If that does not work, please post your strace log

deepakdc01, 2021-08-10:
Hi @haitaohuang, it was the same result when I checked before.
Anyways, it looks like the problem has disappeared after I restarted the host after filing this bug report. The aesmd service runs OK and the sample code are able to work as desired.

closed 2021-08-10 by deepakdc01