intel/linux-sgx-driver #141: About the driver's stability through reboots

p-vernaeckt, 2021-09-30:
Hi.

I use the SGX drivers on many Intel NUCs, some of them commute with me and thus have to be shut down/restarted.

I noticed that I often lose `/dev/isgx` upon restart. 

Right now I solve this using Scone's one-liner (which is basically a wrapper for what's inside this repo's README):

```
curl -fssl https://raw.githubusercontent.com/SconeDocs/SH/master/install_sgx_driver.sh | bash
```

But this is clearly not a long-term solution.

So my question is the following: is the SGX driver made to persist across reboots or are these symptoms normal?

My setup:

```
OS: Ubuntu 20.04.3 LTS x86_64
Host: NUC7i7BNH J31153-311 
Kernel: 5.4.0-88-generic 
CPU: Intel i7-7567U (4) @ 4.000GHz 

SGX driver version: master branch from this repo.
```

joman-xmargin, 2021-09-30:
Make sure that the isgx driver is in your /etc/modules.

p-vernaeckt, 2021-10-01:
Yes it is.

ENOTTY, 2021-10-04:
Are your kernels upgrading between reboots? I.e., are you rebooting from one kernel version into another?

Any output in dmesg related to `[i]sgx`?