intel/linux-sgx-driver #143: Unable to install driver: Skipping BTF generation for /opt/intel/sgxdriver/package/isgx.ko due to unavailability of vmlinux

sbellem, 2021-11-03:
I am trying to install the driver (for linux-sgx 2.14) on Ubuntu 20.04, with linux kernel 5.15, and the installation hangs after the lines:

```console
Skipping BTF generation for /opt/intel/sgxdriver/package/isgx.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15-051500-generic'
```

Here's the complete output:
 
```console
$ sudo ./sgx_linux_x64_driver_2.11.0_2d2b795.bin 
Unpacking Intel SGX Driver ... done.
Verifying the integrity of the install package ... done.
Installing Intel SGX Driver ...
/tmp/sgx-driver-rQ6iEY /home/sylvain/Downloads
install -d /opt/intel/sgxdriver/package
install -d /opt/intel/sgxdriver/scripts
install package/* /opt/intel/sgxdriver/package
install scripts/* /opt/intel/sgxdriver/scripts
/home/sylvain/Downloads
/opt/intel/sgxdriver/package /home/sylvain/Downloads
make -C /lib/modules/5.15-051500-generic/build M=/opt/intel/sgxdriver/package modules
make[1]: Entering directory '/usr/src/linux-headers-5.15-051500-generic'
  CC [M]  /opt/intel/sgxdriver/package/sgx_main.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_page_cache.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_ioctl.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_vma.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_util.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_encl.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_encl2.o
  LD [M]  /opt/intel/sgxdriver/package/isgx.o
  MODPOST /opt/intel/sgxdriver/package/Module.symvers
  CC [M]  /opt/intel/sgxdriver/package/isgx.mod.o
  LD [M]  /opt/intel/sgxdriver/package/isgx.ko
  BTF [M] /opt/intel/sgxdriver/package/isgx.ko
Skipping BTF generation for /opt/intel/sgxdriver/package/isgx.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15-051500-generic'

# hanging here
```

sbellem, 2021-11-04:
Solved the issue with the following steps:

```
turn computer off
turn computer on
go into BIOS
disable SGX
boot
turn computer off
go into BIOS
enable SGX
boot
```

sbellem, 2021-11-30:
The trick used above (in https://github.com/intel/linux-sgx-driver/issues/143#issuecomment-960311014) does not work anymore. The in-kernel driver `/dev/sgx` is always there, meanwhile the out-of-tree driver, `/dev/isgx` is not.

sbellem, 2021-12-02:
Output of `dmesg`:

```bash
[  105.882805] isgx: loading out-of-tree module taints kernel.
[  105.882843] isgx: module verification failed: signature and/or required key missing - tainting kernel
[  105.883175] intel_sgx: Intel SGX Driver v2.11.1                                                                                           
[  105.883194] intel_sgx INT0E0C:00: EPC bank 0x50200000-0x55f80000                                                                          
[  105.902148] intel_sgx:  can not reset SGX LE public key hash MSRs                                                                         
[  105.902222] intel_sgx: second initialization call skipped
```

haitaohuang, 2022-03-17:
The error is a warning, installation should work fine. You dmesg indicates it was installed. Not sure why you don't see /dev/isgx
here is the log on my system with FLC (note u should not use this driver on FLC system):
```
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-1017-azure'
  CC [M]  /opt/intel/sgxdriver/package/sgx_main.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_page_cache.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_ioctl.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_vma.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_util.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_encl.o
  CC [M]  /opt/intel/sgxdriver/package/sgx_encl2.o
  LD [M]  /opt/intel/sgxdriver/package/isgx.o
  MODPOST /opt/intel/sgxdriver/package/Module.symvers
  CC [M]  /opt/intel/sgxdriver/package/isgx.mod.o
  LD [M]  /opt/intel/sgxdriver/package/isgx.ko
  BTF [M] /opt/intel/sgxdriver/package/isgx.ko
Skipping BTF generation for /opt/intel/sgxdriver/package/isgx.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-1017-azure'

uninstall.sh script generated in "/opt/intel/sgxdriver".

Installation is successful!
```
output of "ls -al /dev/*sgx*" may help identify why /dev/isgx not showing up.

sbellem, 2022-03-18:
Somewhat related, but for a machine where the kernel driver (`/dev/sgx`) is not installed, and for which the out-of-tree driver does not appear, even thought it was installed. Right now, when this problem happens, I uninstall the driver and re-install it, and then the `dev/isgx` shows up.

haitaohuang, 2022-03-18:
you should see errors in kernel log when you install isgx.ko and /dev/isgx not showing up.

closed 2022-06-13 by haitaohuang