intel/linux-sgx-driver #144: make fails: Skipping BTF generation for ...isgx.ko due to unavailability of vmlinux

sbellem, 2021-12-01:
```console
$ make
make -C /lib/modules/5.15.5-051505-generic/build M=/home/pi/intel/linux-sgx-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.5-051505-generic'
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_main.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_page_cache.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_ioctl.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_vma.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_util.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_encl.o
  CC [M]  /home/pi/intel/linux-sgx-driver/sgx_encl2.o
  LD [M]  /home/pi/intel/linux-sgx-driver/isgx.o
  MODPOST /home/pi/intel/linux-sgx-driver/Module.symvers
  CC [M]  /home/pi/intel/linux-sgx-driver/isgx.mod.o
  LD [M]  /home/pi/intel/linux-sgx-driver/isgx.ko
  BTF [M] /home/pi/intel/linux-sgx-driver/isgx.ko
Skipping BTF generation for /home/pi/intel/linux-sgx-driver/isgx.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.5-051505-generic'
```

anand10717, 2022-02-10:
+1 Facing the same issue.

feknall, 2022-03-13:
Isn't there still any solutions for this problem?

haitaohuang, 2022-03-17:
This is not a fatal error. The driver is still built. However, I notice you are using 5.11+ kernel which should already have sgx built-in. Unless you are using non-FLC, you should not need this driver. Our PSW would automatically use /dev/sgx/enclave if it exists and ignore /dev/isgx.

closed 2022-06-13 by haitaohuang

ericyame, 2023-11-01:
Hi, I have the same issue with kernel 5.11. I don't see any sgx driver under /dev. Also I cannot build sgx driver like the original post. Any comment?

oldoldcoder, 2024-10-28:
+1,can u solve?

andyzyb, 2024-10-28:
kernel 5.11 should have built-in SGX driver, so you don't need to build this driver.