intel/linux-sgx-driver #146: How to add driver module in Linux Kernel 5.17?

graziadonghia, 2022-09-14:
Hello, I have ubuntu 21.10 and my hardware supports FLC. Based on these assumption, I should already have the SGX driver installed in my laptop, right? This is the output of `ls -ltr /dev/*sgx*`:
```
crw------- 1 root root    10, 124 Sep 14 15:37 /dev/sgx_vepc
crw-rw---- 1 root sgx_prv 10, 126 Sep 14 15:37 /dev/sgx_provision
crw-rw---- 1 root sgx     10, 125 Sep 14 15:37 /dev/sgx_enclave

/dev/sgx:
total 0
lrwxrwxrwx 1 root root 16 Sep 14 15:37 provision -> ../sgx_provision
lrwxrwxrwx 1 root root 14 Sep 14 15:37 enclave -> ../sgx_enclave
```
However the driver is not in kernel modules, since it doesn't show any output if I type `lsmod | grep sgx`
How can I enable the module? Where is the `sgx.ko` file? I didn't download the driver because the community suggested me to not do it because I could have conflicts between driver versions.
Thanks in advance

ScottR-Intel, 2022-09-14:
Hi.

Since you already see the /dev/sgx* devices, the module is already loaded and running.  You can also confirm with dmesg:

     sudo dmesg | grep -i sgx

You should see the SGX EPC range printed out.

Regards.

Scott

graziadonghia, 2022-09-14:
Yes, I see it. Thank you very much.

graziadonghia, 2022-09-19:
I have another problem. I installed both SGX SDK and SGX PSW but when I try to run the Sample Code I get this error: 
```
[get_driver_type edmm_utility.cpp:116] Failed to open Intel SGX device.
[get_driver_type /home/grace/thesis/linux-sgx/psw/urts/linux/edmm_utility.cpp:116] Failed to open Intel SGX device.
failed to load enclave.
```
I don't get it, why is the driver not found? should I run `sudo /sbin/depmod` and `sudo /sbin/modprobe sgx` ? In that case this is the output:
```
modprobe: FATAL: Module sgx not found in directory /lib/modules/5.17.5-76051705-generic
```

xiangquanliu, 2022-09-20:
What's the current user? Is it root? If it is not root user, please check if the current user belongs to sgx group. Try to add the current user to sgx group and re-login. Then i think it will work.

graziadonghia, 2022-09-20:
Yes, thank you so much. I don't know why I didn't think about privileges in the first place