intel/linux-sgx-driver #149: Calling enclave_alloc error

alexthomas1, 2023-04-24:
When I try to call enclave_alloc by calling dl_open on libsgx_enclave_common.so.1, I get this error.

[error_driver2api sgx_enclave_common.cpp:266] Enclave not authorized to run, .e.g. provisioning enclave hosted in app without access rights to /dev/sgx_provision. You need add the user id to group sgx_prv or run the app as root.

lzha101, 2023-04-24:
Looks you are not using this OOT driver but the DCAP driver/in-kernel driver.  Please try to add your user id to group sgx_prv with below command: 

> Add the user running the process to the sgx_prv group:
$ sudo usermod -a -G sgx_prv \<user name\>

alexthomas1, 2023-04-24:
That's the first thing I tried actually, that didn't seem to work. To add some more context, I am trying to run this as an OCALL in open-enclave.

lzha101, 2023-04-25:
Not quite sure about your usage. If you are trying to loading an enclave that requiring provision key access, you may get the above error. Then you can try to check whether the device has correct permission: 

> $ ls /dev/sgx* -l
crw-rw-rw- root root      /dev/sgx/enclave
crw-rw---- root sgx_prv   /dev/sgx/provision

If yes, suppose after you add your user id to group sgx_prv, the error should be gone.

alexthomas1, 2023-04-25:
Thanks for the reply. I think the issue is due to enclave_alloc calling:

https://github.com/intel/SGXDataCenterAttestationPrimitives/blob/94e327a8235bc7c85353dc5e0a6dc1564e6c96c6/driver/linux/encl.c

It returns EACCESS quite generally here, which triggers the print statement. I am trying to implement edmm support inside OpenEnclave

lzha101, 2023-04-25:
OK. Are you using the DCAP driver?  In my memory, DCAP driver doesn't support EDMM.  So you need to use in-kernel driver instead for the edmm feature.

alexthomas1, 2023-04-25:
Thank you. How can you tell whether you use the DCAP driver or in-kernel? Here's my output

```
crw-rw-rw- 1 root root    10, 125 Apr 12 14:38 /dev/sgx_enclave
crw-rw---- 1 root sgx_prv 10, 126 Apr 12 14:38 /dev/sgx_provision
crw------- 1 root root    10, 124 Apr 12 14:38 /dev/sgx_vepc

/dev/sgx:
total 0
lrwxrwxrwx 1 root root 14 Apr 12 14:38 enclave -> ../sgx_enclave
lrwxrwxrwx 1 root root 16 Apr 12 14:38 provision -> ../sgx_provision
```

ScottR-Intel, 2023-04-25:
Hi @alexthomas1.

The easiest way to check is using dmesg, ie:

sudo dmesg | grep -i sgx

If you get an output of simply the EPC space, you're using the in-kernel driver, ie:

sudo dmesg | grep -i sgx
[   11.769850] sgx: EPC section 0x4060000000-0x7fff3fefff

If you get more output that is preceded by "intel_sgx:", you're using the DCAP driver, ie:

sudo dmesg | grep -i sgx
[  264.595989] intel_sgx: loading out-of-tree module taints kernel.
[  264.596018] intel_sgx: module verification failed: signature and/or required key missing - tainting kernel
[  264.596219] intel_sgx: EPC section 0x60300000-0x77cfffff
[  264.597220] intel_sgx: Intel SGX DCAP Driver v1.41

FYI, you would have had to download and install the DCAP driver manually.

Regards.

Scott

alexthomas1, 2023-04-25:
Thanks, I am using the in-kernel driver.

ScottR-Intel, 2023-04-25:
Ok, good.  But one more cavate about EDMM + in-kernel driver....  EDMM support wasn't introduced in the in-kernel driver until v6.0.  Please make sure you have at least that version.

uname -a

Can confirm...

Scott

alexthomas1, 2023-04-25:
Yes, I have at least Kernel 6.0. I ran the EDMM tests in the external library and it worked. Currently, I ported all of the EDMM logic (EMA, etc.) in-enclave. I then ran sgx_mm_init on open-enclave's base address and end address; however, I believe sgx_mm_init relies on the guard page layout (?) https://github.com/intel/sgx-emm/blob/08ec21c04047a52eed016e62f077509ab6d650e8/emalloc.c#L426

However it seems like in the most recent commit, sgx_mm_init no longer reserves memory? https://github.com/intel/sgx-emm/commit/08ec21c04047a52eed016e62f077509ab6d650e8

closed 2023-05-03 by alexthomas1