intel/linux-sgx-driver #152: linux-sgx-driver compilation fails when the kernel version is 6.4.

Shuiqing321, 2023-07-24:
As the title suggests, in the sgx_main.c file, there's a setting for the vm_flags variable. The kernel has already made modifications to the attributes of this variable by adding 'const' and provided an interface for configuration.
the kernel commit：https://github.com/torvalds/linux/commit/bc292ab00f6c7a661a8a605c714e8a148f629ef6
[image]

joshwyant, 2024-04-27:
It looks like the latest driver is also failing to install for me.

I'm working through the Driver Installation QuickStart instructions in the latest [Intel SGX SW Installation Guide for Linux](https://download.01.org/intel-sgx/latest/linux-latest/docs/Intel_SGX_SW_Installation_Guide_for_Linux.pdf). I'm following the recommended instructions for the Out-of-Tree Driver option, as I need Legacy Launch Control support for my machine.

I'm unable to install the driver, and I've tried on Ubuntu 22.04.4 and 24.04 Desktop.

According to the [readme](https://download.01.org/intel-sgx/latest/linux-latest/distro/driver_readme.txt) I found, the OOT version is `sgx_linux_x64_driver_2.11.b6f5b4a.bin`.

Here's what I get when I try to install:

```
$ sudo ./sgx_linux_x64_driver_2.11.b6f5b4a.bin 
Unpacking Intel SGX Driver ... done.
Verifying the integrity of the install package ... done.
Installing Intel SGX Driver ...
/tmp/sgx-driver-ciUmNZ /home/josh
install -d /opt/intel/sgxdriver/package
install -d /opt/intel/sgxdriver/scripts
install package/* /opt/intel/sgxdriver/package
install scripts/* /opt/intel/sgxdriver/scripts
/home/josh
Warning: There is no need to install additional SGX driver with in-kernel SGX support.
/opt/intel/sgxdriver/package /home/josh
make -C /lib/modules/6.5.0-28-generic/build M=/opt/intel/sgxdriver/package modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-28-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /opt/intel/sgxdriver/package/sgx_main.o
/opt/intel/sgxdriver/package/sgx_main.c: In function ‘sgx_mmap’:
/opt/intel/sgxdriver/package/sgx_main.c:112:23: error: assignment of read-only member ‘vm_flags’
  112 |         vma->vm_flags |= VM_PFNMAP | VM_DONTEXPAND | VM_DONTDUMP | VM_IO |
      |                       ^~
make[3]: *** [scripts/Makefile.build:251: /opt/intel/sgxdriver/package/sgx_main.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.5.0-28-generic/Makefile:2039: /opt/intel/sgxdriver/package] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-28-generic'
make: *** [Makefile:16: default] Error 2
```

specifically,

```
/opt/intel/sgxdriver/package/sgx_main.c: In function ‘sgx_mmap’:
/opt/intel/sgxdriver/package/sgx_main.c:112:23: error: assignment of read-only member ‘vm_flags’
  112 |         vma->vm_flags |= VM_PFNMAP | VM_DONTEXPAND | VM_DONTDUMP | VM_IO |
      |                       ^~
```

I get the same error on 24.04. I also made sure the prerequisites were installed.

It looks like PR #151 was a good fix, but closed with no explanation; but it was applied as a [patch to a fork](https://github.com/intel/linux-sgx-driver/commit/2f69bc44869a23691001148bce16d26f77ba030a).

Also, the warning, "There is no need to install additional SGX driver with in-kernel SGX support." is misleading if you need to use the legacy /dev/isgx, as it's only available in the OOT version of the sgx driver.

joshwyant, 2024-04-27:
Addresses https://github.com/intel/linux-sgx-driver/issues/152

ottacom, 2025-02-10:
I simply replaced that row 112 with:

vm_flags_set(vma, VM_PFNMAP | VM_DONTEXPAND | VM_DONTDUMP | VM_IO | VM_DONTCOPY);

and it works for me on ubuntu 24.04 noble.

It's something related with the new kernel and the vm_flags