intel/linux-sgx-driver #156: run ./sgx_linux_x64_driver_1.36.2.bin  error

xiaoran33, 2024-01-12:
run ./sgx_linux_x64_driver_1.36.2.bin  error:
```
Unpacking Intel SGX Driver ... done.
Verifying the integrity of the install package ... done.
Installing Intel SGX Driver ...
/tmp/sgx-driver-8gBXzL ~
install -d /opt/intel/sgxdriver/package
install -d /opt/intel/sgxdriver/scripts
cp -r package/* /opt/intel/sgxdriver/package
install scripts/* /opt/intel/sgxdriver/scripts
~

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KDIR=/lib/modules/5.15.0-91-generic/build....(bad exit status: 2)
ERROR (dkms apport): binary package for sgx: 1.36.2 not found
Error! Bad return status for module build on kernel: 5.15.0-91-generic (x86_64)
Consult /var/lib/dkms/sgx/1.36.2/build/make.log for more information.
```
================================================
```
make.log info:
DKMS make.log for sgx-1.36.2 for kernel 5.15.0-91-generic (x86_64)
2024年 01月 12日 星期五 14:39:41 CST
make -C /lib/modules/5.15.0-91-generic/build M=/var/lib/dkms/sgx/1.36.2/build CFLAGS_MODULE="-I/var/lib/dkms/sgx/1.36.2/build -I/var/lib/dkms/sgx/1.36.2/build/include" modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-91-generic'
  CC [M]  /var/lib/dkms/sgx/1.36.2/build/encl.o
  CC [M]  /var/lib/dkms/sgx/1.36.2/build/main.o
/var/lib/dkms/sgx/1.36.2/build/main.c: In function ‘sgx_init’:
/var/lib/dkms/sgx/1.36.2/build/main.c:792:9: note: #pragma message: kernel version may not be supported
  792 | #pragma message "kernel version may not be supported"
      |         ^~~~~~~
  CC [M]  /var/lib/dkms/sgx/1.36.2/build/driver.o
  CC [M]  /var/lib/dkms/sgx/1.36.2/build/ioctl.o
  LD [M]  /var/lib/dkms/sgx/1.36.2/build/intel_sgx.o
  MODPOST /var/lib/dkms/sgx/1.36.2/build/Module.symvers
ERROR: modpost: "kallsyms_lookup_name" [/var/lib/dkms/sgx/1.36.2/build/intel_sgx.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /var/lib/dkms/sgx/1.36.2/build/Module.symvers] Error 1
make[2]: *** Deleting file '/var/lib/dkms/sgx/1.36.2/build/Module.symvers'
make[1]: *** [Makefile:1830: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-91-generic'
make: *** [Makefile:22: default] Error 2
```

How to solve it??

jasonplato, 2024-03-25:
same question

KoraXia, 2024-07-18:
> ERROR: modpost: "kallsyms_lookup_name" [/var/lib/dkms/sgx/1.36.2/build/intel_sgx.ko] undefined!

According to: https://github.com/xcellerator/linux_kernel_hacking/issues/3, kallsyms_lookup_name is not exported anymore in kernels > 5.7
So I solved it by downgrading the kernel to version 5.7 or lower (I use 5.4).

jgnoonan, 2024-12-08:
I understand that this driver is no longer needed if you have kernel version 5.11 or higher.  The SGX driver is now built into the kernel.  If you look in /dev and you see sgx_provision and sgx_enclave, you likely do not need the driver file.