intel/linux-sgx-driver #160: Error in sgx_main.c - Assignment of Read-Only Member vm_flags on Branch 2.14

DylanCkawalec, 2024-05-24:
#### Description
When attempting to build the Intel SGX driver on a system with kernel version 6.5.0-1021-azure, the following error occurs:
```
/home/username/linux-sgx-driver/sgx_main.c: In function ‘sgx_mmap’:
/home/username/linux-sgx-driver/sgx_main.c:112:23: error: assignment of read-only member ‘vm_flags’
  112 |         vma->vm_flags |= VM_PFNMAP | VM_DONTEXPAND | VM_DONTDUMP | VM_IO |
      |                       ^~
make[3]: *** [scripts/Makefile.build:251: /home/username/linux-sgx-driver/sgx_main.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.5.0-1021-azure/Makefile:2039: /home/username/linux-sgx-driver] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-1021-azure'
make: *** [Makefile:16: default] Error 2
```

#### Cause
The error is caused by the code attempting to modify the `vm_flags` field of the `vma` structure directly, which is marked as read-only in recent kernel versions.

#### Solution
To resolve this issue, modify the `sgx_main.c` file to use an indirect method for modifying the `vm_flags` field. Here’s the corrected code snippet:

```c
static int sgx_mmap(struct file *file, struct vm_area_struct *vma)
{
    vma->vm_ops = &sgx_vm_ops;
    unsigned long new_flags = vma->vm_flags | VM_PFNMAP | VM_DONTEXPAND | VM_DONTDUMP | VM_IO | VM_DONTCOPY;
    *(unsigned long *)&vma->vm_flags = new_flags;
    return 0;
}
```

#### Steps to Reproduce
1. Clone the Intel SGX driver repository.
2. Attempt to build the driver using `make` on a system with kernel version 6.5.0-1021-azure.
3. Observe the compilation error related to the `vm_flags` field in `sgx_main.c`.

#### Expected Behavior
The driver should compile without errors.

#### Environment
- Kernel Version: 6.5.0-1021-azure
- GCC Version: 11.4.0
- Intel SGX Driver Version: 2.14.0

#### Additional Context
This issue and its solution were discussed and resolved during a development process. The fix involves using a safer approach to modify the read-only `vm_flags` field.

GermanAizek, 2024-11-13:
Same error on master branch, commit https://github.com/intel/linux-sgx-driver/commit/bd80d04d52d818c2f8690daa9dcd676ddb5c1af4

GermanAizek, 2024-11-13:
I found solution on https://github.com/Jumpst3r/linux-sgx-driver/commit/215313bb638f87cd2aa3b6e9284c23eba286c214